jjshoots/PyFlyt #1: For some reason simulation always runs at 0.5x realtime speed

jjshoots, 2022-12-07:


rickstaa, 2023-07-24:
Placing this here for future reference.

> @jjshoots I found out about this change because [eb85518](https://github.com/jjshoots/PyFlyt/commit/eb855187974637ffecadc9ad8f653c5f2ba000b4) broke the tests on my upstream (see https://github.com/rickstaa/stable-gym/actions/runs/5646243701/job/15293706729). I now fixed it there but I was wondering why it was added thinking. I checked your codebase, and I could not find the reason. Pybullet returns a tuple instead of a numpy array if numpy acceleration is not enabled. The only downside is that it may degrade performance. Is it possible to throw a warning instead?
> 
> ```python
> PYBULLET_NUMPY_WARNING_THROWN = False
> 
> _, _, rgbaImg, depthImg, segImg = self.p.getCameraImage(
>     height=self.camera_resolution[0],
>     width=self.camera_resolution[1],
>     viewMatrix=self.view_mat,
>     projectionMatrix=self.proj_mat,
> )
> 
> # Convert to numpy arrays if pybullet was not built with numpy support.
> if not pybullet.isNumpyEnabled():
>     rgbaImg, depthImg, segImg = (
>         np.asarray(rgbaImg),
>         np.asarray(depthImg),
>         np.asarray(segImg),
>     )
> 
>     # Warn the user that pybullet was not built with numpy support.
>     if not PYBULLET_NUMPY_WARNING_THROWN:
>         logging.warning(
>             "PyBullet is not installed properly with Numpy functionality. "
>             "This will result in a significant performance hit.\n"
>             "To fix this install Numpy again and following rebuild PyBullet:\n"
>             "\tpip3 uninstall pybullet -y\n"
>             "\tpip3 install numpy\n"
>             "\tpip3 install pybullet --no-cache-dir\n"
>         )
>         PYBULLET_NUMPY_WARNING_THROWN = True
> ```
> 
> I see that you already convert this tuple to a numpy array in most of your code:
> 
> https://github.com/jjshoots/PyFlyt/blob/ea5b838355b6f8335ba08f83821aaf545a98327c/PyFlyt/gym_envs/fixedwing_envs/fixedwing_base_env.py#L281-L283
> 
> https://github.com/jjshoots/PyFlyt/blob/ea5b838355b6f8335ba08f83821aaf545a98327c/PyFlyt/gym_envs/quadx_envs/quadx_base_env.py#L275-L277
> 
> The only place this is not done is:
> 
> https://github.com/jjshoots/PyFlyt/blob/abd3315d60250336a3c29396161af2aaf9583c8d/PyFlyt/core/abstractions/camera.py#L158-L165
> 
> https://github.com/jjshoots/PyFlyt/blob/ea5b838355b6f8335ba08f83821aaf545a98327c/PyFlyt/gym_envs/rocket_envs/rocket_base_env.py#L322-L324
> *Originally posted in https://github.com/jjshoots/PyFlyt/pull/8#issuecomment-1648439585*

rickstaa, 2023-07-24:
Not too much of a problem since I have now fixed it in my upstream (see https://github.com/rickstaa/stable-gym/pull/254/files) 👍🏻.

jjshoots, 2023-07-25:
@rickstaa BTW, stable-gym looks pretty cool, and I'm curious to know more, I don't mind contributing either. However, I couldn't find a definition to "positive definite reward function", could you link me to a relevant place where I can read more?

rickstaa, 2023-07-25:
> @rickstaa BTW, stable-gym looks pretty cool, and I'm curious to know more; I don't mind contributing, either.

@jjshoots, thanks for the kind word 😄. I'm trying to extend the theory of [Han et al. 2020](https://arxiv.org/abs/2004.14288), who applied Lyapunov control theory to achieve Model-free RL algorithms to train stable algorithms in mean costs. I chose your environments since I thought the added stability might benefit drones. Thanks again for implementing them! I added a new tracking environment and changed the cost function of the `Waypoints` and `Hover` environments to be Positive (semi)-definite. Feel free to create any pull requests to [stable-gym](https://github.com/rickstaa/stable-gym/pull/254/files), [stable-learning-control](https://github.com/rickstaa/stable-learning-control) or ros-gazebo-gym (not yet released) if you see something that can be improved 🙏🏻.

rickstaa, 2023-07-25:
If you would like to be referred to differently in the documentation, feel free to create a PR. You are now referred to by package name inside the [robotics environments](https://rickstaa.dev/stable-gym/envs/envs.html#robotics-environment) and the individual quadrotor environments.

jjshoots, 2023-07-25:
@rickstaa I see that the pyflyt environments within stable-gym have the 'not well tested' warning. Would you like some help with that? I could help you run many runs on them to verify their learnability using an implementation of SAC I have on hand.

rickstaa, 2023-07-25:
> @rickstaa I see that the pyflyt environments within stable-gym have the 'not well tested' warning. Would you like some help with that? I could help you run many runs on them to verify their learnability using an implementation of SAC I have on hand.

They are still marked as `note well tested` because I didn't yet have time to train an RL agent on them 😅. The next items on my backlog are reproducing the results of [Han et al. 2020](https://arxiv.org/abs/2004.14288) and ensuring LAC and SAC agents can train on my new Panda ROS Gymnasium environment. If you have the time and resources to do that, that would be great 🚀!

I was planning to add the training results for [LAC](https://rickstaa.dev/stable-learning-control/usage/algorithms/lac.html) and my implementation of [SAC](https://rickstaa.dev/stable-learning-control/usage/algorithms/sac.html) to https://rickstaa.dev/stable-learning-control/usage/benchmarks.html, but we can also attach some benchmarks to the [stable-gym docs](https://rickstaa.dev/stable-gym/) 🤔. The variant of [SAC](https://rickstaa.dev/stable-learning-control/usage/algorithms/sac.html) I use is similar to the one in [spinningup](https://spinningup.openai.com/en/latest/algorithms/sac.html) and stable-baselines3](https://stable-baselines3.readthedocs.io/en/master/modules/sac.html), but the environment should work with any SAC implementation which minimized the cost.

jjshoots, 2023-07-26:
@rickstaa I'm running experiments on the stable-gym quadX environments now. The results can be viewed here for the hover one: https://wandb.ai/jjshoots/ccge2_stable_gym/sweeps/ortymard/overview?workspace=

rickstaa, 2023-07-27:
> @rickstaa I'm running experiments on the stable-gym quadX environments now. The results can be viewed here for the hover one: https://wandb.ai/jjshoots/ccge2_stable_gym/sweeps/ortymard?workspace=user-
> 
> Not sure what a good total reward score would be, but I reversed the sign of the cost (so it's now a reward) for my RL implementation. Therefore, `total_reward = -total_cost`.

Amazing. I love that you use WAB. On most of my environments, I got between 30-200, so the end reward of around -100 is good. How does the trained policy behave? Is it able to hold the drone still? 

Which architecture/setup are you using to perform the training rollouts? I'm looking for a fast way to perform my experiments. I was looking at Google Cloud, but maybe you have a suggestion 🤓.

jjshoots, 2023-07-27:
@rickstaa I love W&B! I haven't checked what the trained policy behaves like yet. I'm using [my own](https://github.com/jjshoots/E2SAC/blob/ccge2_stable_gym/src/mainSAC.py) implementation of SAC using [an automation tool](https://github.com/jjshoots/Wingman) that integrates W&B seamlessly (it handles names, weights versioning, tracking, configuration). I currentluy run it using tmux on a few machines I have available to me, an example of how I initialize all the runs is [here](https://github.com/jjshoots/E2SAC/blob/ccge2_stable_gym/run.sh). Hope this helps!

rickstaa, 2023-08-08:
> @rickstaa I love W&B! I haven't checked what the trained policy behaves like yet. I'm using [my own](https://github.com/jjshoots/E2SAC/blob/ccge2_stable_gym/src/mainSAC.py) implementation of SAC [...]

Great, thanks for all the tips! I just added W&B to my RL package (see https://github.com/rickstaa/stable-learning-control/releases/tag/v4.2.0). It was straightforward and improved my workflow 🚀.

jjshoots, 2024-04-30:
Fixed from #32

closed 2024-04-30 by jjshoots