jjshoots/PyFlyt #11: Quad waypoints

DaviGuanabara, 2023-08-28:
I am not sure why but I am not able to run the quadx waypoints environment with stable baselines 3. It seems to be a incompatible type (sb3 os requiring gymnasium environment, and, although quadx waypoint parent heirs Gymnasium Env, it is not accepted).
I just assumed that you used sb3. I did not find this implementation here. Could you help me ?

my main goal is to run quadx waypoints environment with drone set_mode(6) 

thank you.

jjshoots, 2023-08-28:
@DaviGuanabara you should be able to wrap the environment in an `ObservationWrapper` from `gymnasium` to be compatible with SB3 like so:

```python
import gymnasium as gym
import numpy as np
import PyFlyt.gym_envs
from gymnasium import ObservationWrapper
from gymnasium.spaces import Box

class FlatQuadXWaypoints(ObservationWrapper):
    def __init__(self, context_length=2):
        super().__init__(env=gym.make("PyFlyt/QuadX-Waypoints-v0"))
        self.context_length = context_length
        attitude_shape = self.env.observation_space["attitude"].shape[0]
        target_shape = self.env.observation_space["target_deltas"].feature_space.shape[
            0
        ]

        self.observation_space = Box(
            low=-np.inf, high=np.inf, shape=(attitude_shape + target_shape,)
        )

    def observation(self, obs):
        new_obs = np.concatenate(
            [obs["attitude"], *[obs["target_deltas"][i] for i in self.context_length]]
        )
        return new_obs

env = FlatQuadXWaypoints()
```

Let me know if you encounter any difficulties with this.

jjshoots, 2023-08-28:
Update, I have released PyFlyt 0.12.0 which allows you to flatten the waypoints environment into something compatible with SB3 using a built-in wrapper. See the docs [here](https://jjshoots.github.io/PyFlyt/documentation/gym_envs/quadx_envs/quadx_waypoints_env.html#flattening-the-environment).

closed 2023-09-20 by jjshoots