jjshoots/PyFlyt #14: Using pwm as actions in the environment

KitStandart, 2023-09-20:
Hi! I want to use an action in the environment in the form of PWM, that is, that I can pass the action = np.array(shape(4),) in the env.step() method. QuadXBaseEnv environment. Should I set env.set_mode(-1) or (7) ? And what dimensions of actions will there be?

jjshoots, 2023-09-20:
Hi! Thanks for the interest! You would be using `env.set_mode(-1)` for direct PWM commands. The list of modes for QuadX are available [here](https://jjshoots.github.io/PyFlyt/documentation/core/drones/quadx.html#PyFlyt.core.drones.QuadX.set_mode). M1-4 follow the [convention defined in Ardupilot/PX4](https://ardupilot.org/copter/docs/connect-escs-and-motors.html#quadcopter) for the Quad X configuration.

The dimensions of the actions are `(4,)` in the range of [0, 1].

jjshoots, 2023-09-20:
Sure! All frames are essentially FLU (Front Left Up), so X points out the nose of the aircraft, Y points out the left, and Z points vertically upwards. All rotation follows [standard right hand grip rules](https://en.wikipedia.org/wiki/Right-hand_rule#:~:text=A%20rotating%20body,-Conventional%20direction%20of&text=The%20length%20of%20the%20vector,positive%20direction%20of%20the%20axis.).

Essentially:
- -1: motors 1 through 4 accordingn to Ardupilot/PX4 conventions, in range of [0, 1]
- 0: [roll rate, pitch rate, yaw rate , thrust]
- 1: [roll angle, pitch angle, yaw angle, climb rate]
- 2: [roll rate, pitch rate, yaw rate , altitude] (this is a degenerate controller and has non-feasible regimes)
- 3: [roll angle, pitch angle, yaw angle , altitude] (this is also a degenerate controller if you don't limit the angles)
- 4: [body frame x velocity, body frame y velocity, yaw rate, altitude]
- 5: [body frame x velocity, body frame y velocity, yaw rate, climb rate]
- 6: [ground frame x velocity, ground frame y velocity, yaw rate, climb rate]
- 7: [ground frame x position, ground frame y position, yaw angle, altitude]

Some notes:
- All linear velocities are in m/s, rotational rates are in radians per second
- All positions are in meters, angular positions in radians
- Ground frame refers to the frame that's stuck to the world `xyz=0,0,0` and does not move
- Body frame refers to a frame that's stuck on the drone, with the z-axis always pointing upwards while xy is rotated according to the drone's yaw angle to keep the x-axis pointing forward out of the drone

closed 2023-09-20 by KitStandart