jjshoots/PyFlyt #16: Support custom environments

wagh311, 2023-11-15:
great job!I would like to ask, does the project support custom environments? For example, changing the simulation environment, adding obstacles, etc.?Thanks.

jjshoots, 2023-11-15:
Of course! That was the intended goal all along. Feel free to look at any of the Gymnasium environments code to see how you can make your own. I am also happy to prpovide assistance and guidance if you have a starting point and an end goal in mind.

wagh311, 2023-11-15:
Thank you for your reply. My current job is to use MADRL to enable multiple drones to track moving ground targets in a fixed-size area. Specifically, there are several static obstacles in the mission area, and the UAV needs to avoid obstacles and continuously track the target based on the image information obtained by the camera. Do you think I can accomplish this with the help of PyFlyt? Thanks again!

jjshoots, 2023-11-15:
Yep, that's definitely possible with PyFlyt. I already do something [similar](https://github.com/jjshoots/pyflyt_rail_env) - flying a UAV along a railway while avoiding obstacles along the way. Beyond that, I think an enviroment that you can look at is the QuadX Gates environment. It's not part of the official PyFlyt package since I haven't had the time to ensure that it's learnable yet. But I think you can accomplish what you need by bootstrapping off that environment.

wagh311, 2023-11-16:
That's really cool, I'll give it a try. Thanks!

closed 2023-11-16 by wagh311

wagh311, 2023-11-17:
OK, thank you very much for your advice!