jjshoots/PyFlyt #17: Quad-X Propellers Axis of Rotation

mchawa, 2023-12-26:
Hello,

From what I can see the body frame is following FLU (Front Left Up) orientation, and the QuadX motors are configured as so:
Motor 0: Front Right
Motor 1: Back Left
Motor 2: Front Left
Motor 3: Back Right

quadx.py line 64-75
```python
            """
            DRONE CONTROL
            motor ids correspond to quadrotor X in PX4,
            -------------------------
            |  (cw)2  0(ccw)        |
            |      \\//           x |
            |      //\\         y_| |
            | (ccw)1  3(cw)         |
            -------------------------
            using the ENU convention
            control commands are in the form of roll-pitch-yaw-thrust
        """
``` 

however the torque coefficient is written as so:
```python
            torque_coef = np.array(
                [
                    -motor_params["torque_coef"],
                    -motor_params["torque_coef"],
                    +motor_params["torque_coef"],
                    +motor_params["torque_coef"],
                ]
            )
``` 

if I understand correctly, I think is is a mistake as Motors 0,1 should have a +ve torque_coeff (CCW about Z-Axis pointing up) and Motors 2,3 should have a -ve torque_coef (CW about Z-Axis pointing up)

mchawa, 2023-12-26:
Nvm , Torque is in opposite direction of rotation. Closed.

closed 2023-12-26 by mchawa

jjshoots, 2023-12-26:
Glad you got this figured out! :)