jjshoots/PyFlyt #18: Proposal: Add NED Interial frame and FRD Body frame

mchawa, 2023-12-28:
I need to have the inertial frame follow NED and body frame follow FRD.

I already did a good progress in this, I was wondering if you have a preferred method to do so to add it to the framework.

Why I need this ? To facilitate the integration into an Ardupilot custom controller.

jjshoots, 2023-12-30:
Apologies for the slow reply, this looks like a great idea. I'm not exactly sure what the best way to do this is that allows for speed of execution, so I'm open to suggestions.

At first glance I think this would work well as a wrapper that wraps the core Aviary class, though this would break downstream environments if used in conjunction.

Happy to see what you have in mind.

mchawa, 2024-01-02:
Here are the following points of differences I found so far:

In Aviary.py
1. Gravity Direction : Must be 9.81 not -9.81
2. Camera Position : A good location is found @ (cameraDistance=5, cameraYaw=135, cameraPitch=50, cameraTargetPosition=[0, 0, 0])
3. Plane loading : I created a new plane

In Quadx.py
1. thrust_coef must be multiplied by -1
2. torque_coef must be multiplied by -1

In Pid.py
Either gains can be changed or PID can be adapted to account for both options.

What I suggest is the following:
1. Add an argument to Aviary named orientation_convention that can takes either "NED_FRD" or "ENU_FLU".
2. Add same argument to DroneClass so that it can be inherited to QuadX or any other drone.
3. In Vehicle's yaml file add an element also named orientation_convention to make sure the urdf is compatible with the selected Convention and throw an error in case there is a mismatch.

jjshoots, 2024-01-02:
I think modifying `Aviary` and all related downstream components to accomodate an additional downstream component is too heavy handed. This adds one more dimension of difficulty when developing using the library. PyFlyt has been designed around composition and KISS, so imho using a wrapper function probably makes more sense.