jjshoots/PyFlyt #21: Comparison with gym-pybullet-drones

Jgunde, 2024-02-01:
How does PyFlyt compare to [gym-pybullet-drones](https://github.com/utiasDSL/gym-pybullet-drones/tree/main)? I'm looking at using one of the libraries to train a quadcopter using reinforcement learning? From what I can tell, in PyFlyt might be easier to create custom quadcopter configurations. Motor ramping is also implemented in PyFlyt but it's still on the [wishlist](https://github.com/utiasDSL/gym-pybullet-drones/tree/main?tab=readme-ov-file#desired-contributionsprs) for gym-pybullet-drones.

jjshoots, 2024-02-01:
Thanks for the question, off the top of my head here are the differences:
- PyFlyt allows you to create more than just quadcopters. We have [fixed wing](https://jjshoots.github.io/PyFlyt/documentation/gym_envs/fixedwing_waypoints_env.html) support, and [rockets](https://jjshoots.github.io/PyFlyt/documentation/gym_envs/rocket_landing_env.html) too.
- PyFlyt supports arbitrary numbers of drones in the environment, with each one being possibly a different type of UAV and possibly operating at different looprates each.
- A [documentation](https://jjshoots.github.io/PyFlyt/documentation.html) website.
- [Wind field](https://jjshoots.github.io/PyFlyt/documentation/core/wind.html) modelling.
- Already has motor dynamics modelling, and has relevant modelling for all dynamics components.
- The UAV model itself has 7 different flight modes to choose from to build an RL agent on top of.

That said, gym-pybullet-drones has the following advantages:
- Betaflight SITL support. Crazyflie SITL support is done through the CrazyFlyt sister library.
- Uses a [unified physics model](https://github.com/utiasDSL/gym-pybullet-drones/blob/main/gym_pybullet_drones/envs/BaseAviary.py#L817), this means it can run faster because it doesn't simulate each motor independently. This is somewhat alleviated by PyFlyt using JIT functions for component calculations, but this is still insufficient to make up the speed difference.

I would use PyFlyt if you're interested in RL research for general UAVs, and hardware support as a secondary function. I would use gym-pybullet-drones if you only want to train for the CrazyFlie UAV and require good hardware support.

Jgunde, 2024-02-01:
Thank you! That was very helpful!
For others reading this, the author's [paper](https://arxiv.org/pdf/2304.01305.pdf) also described the differences between PyFlyt and comparable libraries.

closed 2024-02-01 by Jgunde