jjshoots/PyFlyt #25: Use a different flight mode

team-daniel, 2024-02-18:
In the gym env:

`env = gym.make("PyFlyt/QuadX-Hover-v1")`

how would I change the QuadX drone flightmode to -1?

Thanks :)

jjshoots, 2024-02-18:
Because of the way that Gymnasium is built, there is no obvious way to alter internal attributes which would allow this to happen in the current version of PyFlyt. However, if you have a local instance of PyFlyt cloned and installed from, the relevant line for changing flight modes is [here](https://github.com/jjshoots/PyFlyt/blob/master/PyFlyt/gym_envs/quadx_envs/quadx_base_env.py#L177).

I am open to making this part of the `options` argument if you're willing to create a PR for it. :)

jjshoots, 2024-03-11:
FYI, I've just updated PyFlyt to v0.17.0, which has a `flight_mode` argument for all the environments now. If you'd like, you should use that so you won't have to modify the source and can use PyFlyt natively with the [transform observation wrapper](https://gymnasium.farama.org/api/wrappers/observation_wrappers/#gymnasium.wrappers.TransformObservation) from Gymnasium.

closed 2024-03-13 by jjshoots