jjshoots/PyFlyt #27: Clarification: Motor Negative PWM

mchawa, 2024-03-19:
In the function "_physics_update_" in the file "_motors.py_" it is allowed to pass **PWM** within **[-1, 1]**.

However, in the function "__jitted_compute_thrust_torque_", I'm not sure this is handled correctly as demonstrated below.

```
# rpm to thrust and torque
thrust = (rpm**2) * thrust_coef * thrust_unit
torque = (rpm**2) * torque_coef * thrust_unit
```

The previous lines of codes will convert the negative rpm into a positive rpm thus it will also have the effect of positive thrust.

Also, I'm not sure I get the logical meaning behind it, At first I thought it's to normalize the pwm value.

If someone can clarify this further would be very thankful.

mchawa, 2024-03-19:
This also has been tested with a simulation, I can confirm for a quadcopter with mode '-1' a PWM value of -1 is equivalent to a PWM value of 1.

jjshoots, 2024-03-19:
You're right, in quadx, we [clip all motor saturation values](https://github.com/jjshoots/PyFlyt/blob/master/PyFlyt/core/drones/quadx.py#L442-L450) so it seems that this was not tested.
The appropriate way of fixing motors.py would be:

```python
# rpm to thrust and torque
thrust = (rpm**2) * thrust_coef * thrust_unit * np.sign(rpm)
torque = (rpm**2) * torque_coef * thrust_unit * np.sign(rpm)
```

The reason we allow negative values is to allow the motors to spin in reverse and produce negative thrust.
Would you like to make a PR fixing this? Otherwise I'll do it when

jjshoots, 2024-03-20:
Fixed in latest release (v0.18.0).

closed 2024-03-20 by jjshoots