jjshoots/PyFlyt #29: BUG: QuadX Motor Saturation

mchawa, 2024-03-22:
In the current implementation of the handling of motor saturations in _quadx.py_

It is handled as follows:
```
  # deal with motor saturations
  if (high := np.max(self.pwm)) > 1.0:
      self.pwm /= high
  if (low := np.min(self.pwm)) < 0.05:
      self.pwm += (1.0 - self.pwm) / (1.0 - low) * (0.05 - low)
```

I think this implementation is not robust as it can happen that both the maximum is > 1 and the minimum is less than 0.05 at the same time and it will not be handled correctly.

A more robust implementation I think would be the following:
Have the MAX_PWM and MIN_PWM as parameters set to (1, 0.05) by default.

Use the following helper function instead of the previous lines:
```
@staticmethod
def _motor_saturation(pwm_values: np.ndarray, min_pwm: float, max_pwm: float):
    """
    This function saturates the motors pwm values proportionally to the min_pwm and max_pwm parameters.

    Returns:
        np.array of pwm values saturated between min_pwm and max_pwm.
    """

    max_value = np.max(pwm_values)
    min_value = np.min(pwm_values)

    result = pwm_values

    data_range = max_value - min_value

    if data_range == 0:  # Prevent division by zero if all values are equal
        result = np.clip(pwm_values, min_pwm, max_pwm)
    elif max_value > max_pwm or min_value < min_pwm:
        scale = (max_pwm - min_pwm) / data_range
        result = min_pwm + ((pwm_values - min_value) * scale)

    return result
```

Optionally we can have a boolean parameter HANDLE_MOTOR_SATURATION set to True by default.

I already have this implemented also, if you accept it let me know and I will create the PR.

Thanks.

jjshoots, 2024-03-22:
The original implementation handles cases of lowerbound + upperbound together as well.
Notice that it's two if loops, not a single if-elif.
The first conditional clips handles overbound, the second one clips underbound.
In addition, the original implementation doesn't increase the bounds should either one of them go out of bounds as severely as your proposed implementation.
Re:
```python
from copy import deepcopy
import numpy as np

def handle_new(pwm: np.ndarray) -> np.ndarray:
    max_value = np.max(pwm)
    min_value = np.min(pwm)

    result = pwm

    data_range = max_value - min_value

    if max_value > 1.0 or min_value < 0.05:
        scale = (1.0 - 0.05) / data_range
        result = 0.05 + ((pwm - min_value) * scale)

    return result

def handle_old(pwm: np.ndarray) -> np.ndarray:
    # deal with motor saturations
    if (high := np.max(pwm)) > 1.0:
        pwm /= high
    if (low := np.min(pwm)) < 0.05:
        pwm += (1.0 - pwm) / (1.0 - low) * (0.05 - low)

    return pwm

if __name__ == "__main__":
    for i in range(100):
        input = np.random.random_sample((4,)) * 1.2
        old = handle_old(deepcopy(input))
        new = handle_new(deepcopy(input))
        print(f"Original {input}")
        print(f"Old {old}")
        print(f"New {new}")
        assert np.max(old) < (np.max(input) + 1e-3), f"Old failed. {np.max(old) - np.max(input)}"
        assert np.max(new) < (np.max(input) + 1e-3), f"Old failed. {np.max(new) - np.max(input)}"
        assert np.min(old) > (np.min(input) - 1e-3), f"New failed. {np.min(old) - np.min(input)}"
        assert np.min(new) > (np.min(input) - 1e-3), f"New failed. {np.min(new) - np.min(input)}"

```

jjshoots, 2024-03-22:
Published a new release fixing the saturation issue.

jjshoots, 2024-03-22:
Reopening this, you're right, the current implementation does fail under certain scenarios like multiple values being out of bounds. I'll look into it a bit more.

jjshoots, 2024-03-22:
I think a reasonable solution is:

```python
    # deal with motor saturations
    high, low = np.max(pwm), np.min(pwm)
    pwm_range = (high - low)
    if high > 1.0:
        overflow = max(0.0, low - 1.0)
        pwm -= (high - 1.0) / pwm_range * (pwm - 1.0) + overflow
    high, low = np.max(pwm), np.min(pwm)
    if low < 0.05:
        underflow = max(0.0, 0.05 - high)
        pwm += (0.05 - low) / pwm_range * (high - pwm) + underflow
```

mchawa, 2024-03-23:
Looks good to me ! only if we can make the thresholds configurable would be nice to have.

closed 2024-03-23 by jjshoots