jjshoots/PyFlyt #31: Minor Issues in Wind Model

mchawa, 2024-04-10:
In aviary.py:

The following code:
```
        elif callable(self.wind_type):
            # custom wind field, initialize and check
            self.wind_field = self.wind_type(
                np_random=self.np_random, **self.wind_options
            )
            WindFieldClass._check_wind_field_validity(self.wind_field)
            self.wind_field = self.wind_type(
                np_random=self.np_random, **self.wind_options
            )
```
can be slightly optimized to be:
```
        elif callable(self.wind_type):
            # custom wind field, initialize and check
            self.wind_field = self.wind_type(
                np_random=self.np_random, **self.wind_options
            )
            WindFieldClass._check_wind_field_validity(self.wind_field)
```
as there is no need to re-initialize the wind_field.

Also, in the example inside base_wind_field.py

The following part:
```
        >>>         # but the z velocity varies to the log of height,
        >>>         wind[:, -1] = np.log(position[:, -1]) * self.strength
```
should be modified slightly to be:
```
        >>>         # but the z velocity varies to the log of height,
        >>>         height = np.clip(position[:, -1] + 1, 0, None)
        >>>         wind[:, -1] = np.log(height) * self.strength
```
Where +1 is added so that a height of 0 equals np.log(1) = 0 and avoid np.log(0) = inf.

I also implemented this in a simple concrere wind_field, I can create a pull request with this simple wind_field if you want.

jjshoots, 2024-04-10:
We need to initialize wind field twice because the check actually performs a call on the windfield. If the windfield has any stateful parameters, these would have to be reset. I agree that this is not very optimal and the check can probably be moved to the init instead. Happy to receive a PR for this.

The second example probably wouldn't have to be changed. It only serves as an example and not meant to be physically realistic. It was written for readability.

Thanks for the report!

mchawa, 2024-04-14:
I think a simple solution is the following:
```
            # custom wind field, initialize and check
            WindFieldClass._check_wind_field_validity(self.wind_type(
                np_random=self.np_random, **self.wind_options
            ))
            self.wind_field = self.wind_type(
                np_random=self.np_random, **self.wind_options
            )
```

jjshoots, 2024-04-15:
That's exactly the same thing

closed 2024-04-23 by jjshoots