jjshoots/PyFlyt #32: Bug: Render doesn't simulate time correctly

mchawa, 2024-04-27:
In Aviary.py, I think the following line:

```
self._sim_elapsed += self.update_period * self.updates_per_step
```

should be changed to:

```
self._sim_elapsed += self.update_period
```

or the rendering will not be in sync with the simulation time and RTF won't be calculated correctly.

jjshoots, 2024-04-27:
IIRC, `update_period` is the update time between each physics step, while `updates_per_step` is the number of physics updates per control update. We update the RTF every control step, so we update the `sim_elapsed` every `update_period * updates_per_step`. If this is wrong, could you describe it in more detail for why it's wrong?

mchawa, 2024-04-28:
`update_period` is calculated as so:
```
self.update_period = 1.0 / np.min(all_control_hz)
```

which is not the physics step but rather the simulation step time.

jjshoots, 2024-04-30:
Fixed in https://github.com/jjshoots/PyFlyt/commit/3406dbe651e2141ea62b9052b826a27f76fe1cab

closed 2024-04-30 by jjshoots