jjshoots/PyFlyt #42: Questions abouts simulation time and wall clock time.

ch-MEIJIE, 2024-06-21:
Hi @jjshoots,

Thank you for your excellent work! It has been very helpful for my projects.

As a new user of this repository and PyBullet, I hope my questions are not too stupid.

Q1: When running the gym_env demos, I noticed that the Real-Time Factors (RTFs) are always under 0.5. Does this indicate that the simulation time is about half as slow as the wall clock time? I suspect this might be due to camera rendering, as the RTFs are close to 1 when running simple examples from the examples folder. Additionally, I've tested this on three machines: my MacBook with an M1 CPU shows the lowest RTFs, around 0.05, while my P52 laptop with a P2000 GPU and a workstation with a 3090 GPU both show RTFs around 0.5.

I need better alignment between simulation time and wall clock time because I'm working on an imitation learning project. In imitation learning, it's crucial to collect human demonstration trajectories, generated when humans interact with an environment by sending action instructions at regular intervals. The environment changes its state according to these instructions, allowing humans to observe these changes and generate complete demonstration trajectories. When these trajectories are used in imitation learning algorithms (often involving reinforcement learning processes), ensuring the agent's interaction with the environment mirrors human interaction is essential. If the time progression in the simulation (such as PyBullet) doesn't match real-time (wall clock time), the agent and human may observe different environmental states within the same time interval, even with the same initial state and actions.

Q2: Is it possible to align the simulation time in gym envs with wall clock time? I believe the RTF should be around 1, correct?

Thank you again, and I look forward to your response.

jjshoots, 2024-06-22:
Hi @ch-MEIJIE ! Happy to hear this is helping your research.

Yes, the RTF should be around 1. On your P52 laptop and workstation, does the RTF hover around 0.5 or does it stay at a maximum of 0.5? An RTF of less than 1 for such high end hardware does indeed seem weird.

There was a problem in the past where the delats between each frame wasn't being calculated properly. I'd assume this was fixed [here](https://github.com/jjshoots/PyFlyt/commit/9e28463132600dc88e7a154761da5b8b554a5fe9). Modifying the base Aviary to have no delay between states should mean you can render the environment as fast as possible, seeing if your RTF can indeed be 1.

Aligning the simulation time to real time exactly is not native to the PyBullet physics engine. So we have to resort to sleeps between frames I'm afraid.

jjshoots, 2024-06-22:
@ch-MEIJIE I have just confirmed your issue on my side. In PyFlyt, when [this section](https://github.com/jjshoots/PyFlyt/blob/master/PyFlyt/gym_envs/quadx_envs/quadx_base_env.py#L170) is commented out and the internal camera rendering is disabled, it is possible to get RTF of 1.0. But enabling internal camera rendering seems to be the main bottleneck. I will look into this to see if I can solve it.

jjshoots, 2024-06-22:
After performing some extensive profiling of the code, the entire step function on my machine (Ryzen 7 5800X, 4070 Ti super), I found that the main bottleneck which results in ~0.5 RTF is caused by PyBullet's native camera image renderer, see [here](https://github.com/jjshoots/PyFlyt/blob/master/PyFlyt/core/abstractions/camera.py#L161).

That one function call consumes 16 ms of CPU time, while everything else combined is less than 1 ms. I will look into this a bit further.

ch-MEIJIE, 2024-06-22:
Hi @jjshoots, thank you very much for your in-depth analysis

jjshoots, 2024-06-23:
This sounds like a great addition, I will add this into the repo later in the day. Or would you like to make a PR instead? Making this an argument would work well.

ch-MEIJIE, 2024-06-23:
I will try it.

jet-sony, 2024-06-24:
@ch-MEIJIE PyFlyt 0.22.0 has been released, the change for camera FPS is already implemented. Let me know if you encounter any issues. :)

closed 2024-06-24 by jjshoots