jjshoots/PyFlyt #45: RecursionError: maximum recursion depth exceeded while calling a Python object

defrag-bambino, 2024-06-27:
I am training a PyFlyt gym env using DreamerV3 from the SheepRL library. While doing so, I am running into a `RecursionError: maximum recursion depth exceeded while calling a Python object` error. This is likely because the Bullet env is newly instantiated for every env reset.
Any ideas why this is and how to solve it?

Issue over at SheepRL:
https://github.com/Eclectic-Sheep/sheeprl/issues/306

jjshoots, 2024-06-28:
Thanks for the report! Theoretically the pybullet instance should be cleaned up by garbage collector at every reset, the simplest issue could be when something else is still referencing memory within the env. I'll need to confirm this is the case. I'll look at it later today or tomorrow. Thanks for the report!

jjshoots, 2024-06-28:
I'm currently testing how much memory PyFlyt is using and what the limits for resets are:
```py
import psutil
import gymnasium as gym
from gymnasium.vector import async_vector_env
from PyFlyt import gym_envs

def main() -> None:
    process = psutil.Process()
    env = async_vector_env.AsyncVectorEnv(
        [
            lambda i=i: gym.make("PyFlyt/QuadX-Hover-v2") for i in range(4)
        ]
    )

    i = 0
    while True:
        print(i, process.memory_info().rss)
        env.reset()
        env.step(env.action_space.sample())
        i += 1

if __name__ == "__main__":
    main()
```

AFAICT, there are no issues of memory leaks here. Perhaps `sheeprl` is doing something unorthodox like referencing variables in the env and not releasing the memory for those references. That could cause the infinite recursion error. In any case, I'll be happy to look more into this if you have any more trace logs.

defrag-bambino, 2024-06-28:
Thanks!
The issue seems to be solved in SheepRL when using env.sync=True, which will wait for all envs to finish and close all of them together.

Maybe just out of curiosity: wouldn't it be faster to just reset the pybullet env somehow, instead of stopping and starting it all over again?

jjshoots, 2024-06-28:
Happy to hear that the issue is fixed!

It's fairly debatable whether resetting the env by resetting individual elements is faster. The way we can do that is to track all objects within the environment, then reset all positions and velocities. However, we then run into cases where objects can spawn in and out of the environments and it becomes unclear how to deal with them. There are also objects like waypoints which have no mass and no collision bodies, so it gets more complicated.

In general, the speed of the env is limited by how fast we can make Python calls to Bullet, so the more calls there are the slower it is (see #42 on one specific call taking up 16 ms of time). So as an easy, no gotchas method of reset, PyFlyt just nukes the env and regenerates it, consuming just 2 moderately expensive calls.

defrag-bambino, 2024-06-29:
Interesting. Thanks anyways!

closed 2024-06-29 by defrag-bambino