jjshoots/PyFlyt #52: Question about camera Settings

wagh311, 2024-07-14:
Hello, I have a questions for you about the camera：

If I want to achieve obstacle avoidance with a drone, if I only have a forward-looking camera, does that mean that the speed direction of the drone must be consistent with the direction of the camera? Otherwise, if the drone moves backwards, then it will not be able to obtain the image behind it, will it not be able to avoid obstacles?

Looking forward to your reply~~

jjshoots, 2024-07-15:
In the case of a drone with 1 camera, yes. However, pyflyt actually allows modularity in drone design, so you can very feasibly have a drone with a camera facing in every direction.

jjshoots, 2024-07-15:
[Here](https://github.com/jjshoots/PyFlyt/blob/master/PyFlyt/core/drones/quadx.py#L200) is the section of code where the camera is initialized for the quadx model. You could, for example, add a `camera2` if you desire with its own properties.

wagh311, 2024-07-15:
Thank you for your reply. Due to the requirements of my mission, the drone can only carry one front-facing camera. So how do I limit my speed choices? Should we limit the range of actions directly in the action space, or should we punish unreasonable actions in the reward function? Can you give me some guidance?

jjshoots, 2024-07-15:
I would limit your action space, to only forward directions in this case. Penalizing backward movement doesn't mean that an RL agent won't execute it if it produces a better resul. Another alternative is just using some sort of model with history (like R2D2) and giving the agent the full action space so it can recall past information.

wagh311, 2024-07-16:
Thank you very much for your advice. I'll give it a try。

wagh311, 2024-07-31:
Hi @jjshoots ,I'm sorry to bother you again, but I would like to ask you the following questions:
In my previous question, you suggested that I limit the movement space and only allow forward movement. Since my task is a quad-rotor UAV, the action adopts speed input, that is, vx,vy,vz, and does not include yaw angle control. After further consideration, I believe that the UAV should not only avoid backward movement, but also ensure that the direction of the forward-looking camera is aligned with the speed direction of the UAV, so as to better avoid collision due to lack of access to obstacle information. I wonder if this understanding is correct? If it is correct, how should I modify it in PyFlyt to meet the above conditions?

jjshoots, 2024-08-01:
PyFlyt is currently not built for that purposes out of the box. There would be pretty weird behavior especially when velocity is zero or near zero -
1. Where do we look at zero velocity?
2. Do we need damping for the camera to prevent chaotic angle switching especially at hover?
3. At what speed does the camera start tracking velocity?

This sounds like opening up a can of worms that I don't have the bandwidth to handle at the moment, but I'd be more than happy if you want to mess with it yourself. The relevant section of code is [here](https://github.com/jjshoots/PyFlyt/blob/master/PyFlyt/core/abstractions/camera.py#L122). `use_gimbal` in this case is just an indicator of whether the camera is stabilized for now (pan-tilt 0). You can get the current camera attitude [here](https://github.com/jjshoots/PyFlyt/blob/master/PyFlyt/core/abstractions/camera.py#L109).

wagh311, 2024-08-01:
I'm sorry, I don't understand why you mentioned the drone speed zero situation. Also, can `use_gimbal` help with camera orientation? Airsim before I used, in which `airsim DrivetrainType. ForwardOnly` can guarantee the yaw angular velocity direction all the time, all I want to know whether PyFlyt  have similar function.

jjshoots, 2024-08-01:
PyFlyt hosts two frame of references, a body frame and a ground frame. Ground frame is attached to the ground, and XYZ is always North-West-Up. Body frame is attached to the UAV, and X is always wherever the UAV is facing.

The camera component is linked to the body frame of the UAV. Wherever the UAV points, the camera will point in the same way. This means that, when the UAV flies backwards, the camera will still be looking forwards (opposite direction of travel), unless the UAV physically turns around by yawing 180 degrees.

What you were suggesting is for the camera to follow the linear velocity of the UAV, regardless of UAV orientation (basically following the UAV's ground frame velocity). This is problematic, when the UAV is hovering, the linear velocity will be oscillating between forward, backward, left, and right relative to the ground frame, causing the camera to be constantly swap directions at hover.

Airsim's `forwardonly` for `drivetrain` is not rotating the camera, it is rotating the entire UAV, so that the camera always points in the direction of the UAV. Additionally, they disable the UAV from giving backward velocity commands. If this is what you want, it is easily achievable in PyFlyt. Just set flight mode 5 and disable `u` from being negative values. This will mean that whatever controller you are using for the UAV, it will only be able to send forward velocity setpoints.

Note that only forward velocity setpoints does not imply only forward velocity. When the drone is forced backwards (such as because of wind or collisions), the physics will still cause the UAV to move backwards, with the camera ending up pointing the wrong way in those scenarios.

wagh311, 2024-08-02:
Thank you very much for your detailed explanation. In my mission, I wanted to control a QuadX with a camera to track a moving target by velocity input(vx,vy,vz), and there were obstacles in the environment. Therefore, I am worried that when the direction of the camera is inconsistent with the speed direction of the drone, the obstacle information may not be obtained, resulting in a collision. In Airsim, I through Airsim DrivetrainType. ForwardOnly solved this problem. However, as you explained, there are many differences between Airsim and PyFlyt. So, I'm still confused, how do I solve this problem in PyFlyt? In other words, how do I make sure the drone is always getting the correct information about the obstacles?

jjshoots, 2024-08-02:
I would do flight mode 5, then ensure that `u` > 0, `v` = 0.You can turn by giving yaw rate commands.

wagh311, 2024-08-02:
May I ask,are u,v,w the x,y,z velocities in a body frame?Are vp, vq, vr the angular velocities of pitch,roll,yaw?

jjshoots, 2024-08-02:
No worries about asking.

Yes, uvw are body frame linear velocities.
vp, vq, vr are angular velocities in the body frame.

wagh311, 2024-08-04:
Thanks a lot! You suggest that I turn by giving yaw rate command, may I ask if yaw rate refers to vr? Also, how do I adjust the yaw rate so that the camera is facing in the same direction as the drone's speed?

jjshoots, 2024-08-04:
Here's an example:
```py
"""Spawn a single drone and get its camera image."""
import matplotlib.pyplot as plt
import numpy as np

from PyFlyt.core import Aviary

# the starting position and orientations
start_pos = np.array([[0.0, 0.0, 1.0]])
start_orn = np.array([[0.0, 0.0, 0.0]])

# environment setup
env = Aviary(
    start_pos=start_pos,
    start_orn=start_orn,
    render=True,
    drone_type="quadx",
    drone_options=dict(
        drone_model="primitive_drone",
        use_camera=True,
        camera_fps=30,
    ),
)

# set to velocity control
env.set_mode(5)

# simulate for 1000 steps (1000/120 ~= 8 seconds)
for i in range(999_999):
    if i < 2000:
        # for the first 2000 steps,
        # u=1.0, v=0.0, yaw_rate=0.4, a bit of climb rate so the drone maintains a reasonable height
        env.set_setpoint(0, np.array([1.0, 0.0, 0.4, 0.02]))
    else:
        # for the remaining steps, change directions
        # u=1.0, v=0.0, yaw_rate=-0.4, a bit of climb rate so the drone maintains a reasonable height
        env.set_setpoint(0, np.array([1.0, 0.0, -0.4, 0.02]))
    env.step()

```

wagh311, 2024-08-04:
Can you explain to me why yaw_rate is set at 0.4 or -0.4? In other words, what relationship do u and vr need to meet so that the direction of the camera is always consistent with the direction of the drone speed?

jjshoots, 2024-08-04:
There's is no requirement for yaw rate (`vr`) to be coupled with forward velocity (`u`). In the example above, 0.4 and -0.4 are just example values indicating a yaw rate of 0.4 radian per second in either directions. You can use any value you want (within reason, you probably don't want to use very big values like > 3).

wagh311, 2024-08-05:
Thank you very much!

closed 2024-08-05 by wagh311