jjshoots/PyFlyt #56: Sure! Here's my current working setup: https://github.com/jjshoots/RLTeamAerialCombat/blob/main/src/vec_env_main.py

abdul-mannan-khan, 2024-07-27:
Hello @jjshooots, 

Sorry to opening up the issue again. I tried to run your code `main.py` and I am getting following message 

```
wingman: --------------𓆩𓆪--------------
wingman: Using device cuda:0
wingman: Saving weights to weights/509088...
wingman: New training instance detected, generating weights directory in 3 seconds...
```
Could you please let me know how to run this code? Thank you for your help?

              Sure! Here's my current working setup: https://github.com/jjshoots/RLTeamAerialCombat/blob/main/src/vec_env_main.py

That said, if you're using the gymnasium environments, I would highly recommend something more standard like CleanRL.

_Originally posted by @jjshoots in https://github.com/jjshoots/PyFlyt/issues/55#issuecomment-2251776598_

jjshoots, 2024-07-29:
You gotta wait a few mins after running to actually see results. AFAICT, the script is working correctly. That said, I really recommend not using my code and using something like CleanRL (or even SB3 which is probably simpler to use). My code is experimental which I use to test all sorts of things.

closed 2024-07-30 by jjshoots

abdul-mannan-khan, 2024-07-30:
Thank you so much @jjshoots  for your kind response and putting me in the right directions. Have no words to express my gratitude. Means a lot for me.