jjshoots/PyFlyt #57: Drone with suspended rope?

defrag-bambino, 2024-08-02:
Hi,

I am interested in creating an environment where there is a rope attached to the drone and it is potentially carrying a payload. Similar to what is described in [this research paper](https://www.mdpi.com/2504-446X/8/2/35).
Do you have an idea of how to realize this with PyBullet? I have not found any established solutions on google yet.

Thanks

[image]

jjshoots, 2024-08-03:
That seems like a fun problem! If you are willing to add this to the repo, it would be a great addition.

To answer your question, you can probably define the rope as a series of links. I would define a URDF file for the rope and weight together. Then when you spawn in the drone, you can use PyBullet's set constraint to attach the rope to the drone.

I will post some boilerplate example code here later in the day.

jjshoots, 2024-08-04:
@defrag-bambino It took a whille before I had the time to do this, but here is an example:

**ball_and_string.urdf**
```xml
<?xml version="1.0"?>
<robot name="ball_and_string">

  <!-- Base link (fixed in space) -->
  <link name="base_link">
    <visual>
      <geometry>
        <box size="0.01 0.01 0.01"/>
      </geometry>
    </visual>
    <collision>
      <geometry>
        <box size="0.01 0.01 0.01"/>
      </geometry>
    </collision>
    <inertial>
      <mass value="0.05"/>
      <inertia ixx="0.001" ixy="0.0" ixz="0.0" iyy="0.001" iyz="0.0" izz="0.001"/>
    </inertial>
  </link>

  <!-- Link 1 -->
  <link name="link_1">
    <visual>
      <geometry>
        <cylinder length="0.05" radius="0.01"/>
      </geometry>
    </visual>
    <collision>
      <geometry>
        <cylinder length="0.05" radius="0.01"/>
      </geometry>
    </collision>
    <inertial>
      <mass value="0.01"/>
      <inertia ixx="0.00001" ixy="0.0" ixz="0.0" iyy="0.00001" iyz="0.0" izz="0.00001"/>
    </inertial>
  </link>
  <joint name="joint_base_link_1" type="continuous">
    <parent link="base_link"/>
    <child link="link_1"/>
    <origin xyz="0 0 -0.05" rpy="0 0 0"/>
    <axis xyz="1 0 0"/>
    <dynamics damping="0.000001" friction="0.000001"/>
  </joint>

  <!-- Link 2 -->
  <link name="link_2">
    <visual>
      <geometry>
        <cylinder length="0.05" radius="0.01"/>
      </geometry>
    </visual>
    <collision>
      <geometry>
        <cylinder length="0.05" radius="0.01"/>
      </geometry>
    </collision>
    <inertial>
      <mass value="0.01"/>
      <inertia ixx="0.00001" ixy="0.0" ixz="0.0" iyy="0.00001" iyz="0.0" izz="0.00001"/>
    </inertial>
  </link>
  <joint name="joint_link_1_link_2" type="continuous">
    <parent link="link_1"/>
    <child link="link_2"/>
    <origin xyz="0 0 -0.1" rpy="0 0 0"/>
    <axis xyz="0 1 0"/>
    <dynamics damping="0.000001" friction="0.000001"/>
  </joint>

  <!-- Ball link -->
  <link name="ball">
    <visual>
      <geometry>
        <sphere radius="0.05"/>
      </geometry>
      <material name="black">
        <color rgba="0.0 0.0 0.0 1.0"/>
      </material>
    </visual>
    <collision>
      <geometry>
        <sphere radius="0.05"/>
      </geometry>
    </collision>
    <inertial>
      <mass value="0.1"/>
      <inertia ixx="0.0001" ixy="0.0" ixz="0.0" iyy="0.0001" iyz="0.0" izz="0.0001"/>
    </inertial>
  </link>
  <joint name="joint_link_last_ball" type="continuous">
    <parent link="link_2"/> <!-- Change to the last link name if more links are added -->
    <child link="ball"/>
    <origin xyz="0 0 -0.1" rpy="0 0 0"/>
    <axis xyz="1 0 0"/>
    <dynamics damping="0.000001" friction="0.000001"/>
  </joint>

</robot>
```

**main.py**
```py
import numpy as np

from PyFlyt.core import Aviary

# the starting position and orientations
start_pos = np.array([[0.0, 0.0, 3.0]])
start_orn = np.array([[0.0, 0.0, 0.0]])

# environment setup
env = Aviary(
    start_pos=start_pos,
    start_orn=start_orn,
    render=True,
    drone_type="quadx",
    drone_options=dict(
        drone_model="primitive_drone",
        use_camera=True,
        camera_fps=30,
    ),
)

# spawn in the pendulum
pendulum_id = env.loadURDF(
    "./ball_and_string.urdf",
    basePosition=np.array([0.0, 0.0, 2.9]),
    useFixedBase=False,
    globalScaling=1.0,
)

# register all new bodies with Aviary
env.register_all_new_bodies()

# create a constraint between the base of the pendulum object and the drone
env.createConstraint(
    env.drones[0].Id,
    -1,
    pendulum_id,
    -1,
    env.JOINT_POINT2POINT,
    np.array([0.0, 0.0, 0.0]),
    np.array([0.0, 0.0, 0.0]),
    np.array([0.0, 0.0, 0.1]),
)

# simulate
env.set_mode(7)
env.set_setpoint(0, np.array([0.0, 0.0, 0.0, 5.0]))
for i in range(5000):
    env.step()

```

defrag-bambino, 2024-08-05:
Thanks!
Right now, the rope seems kind of stiff. Even when allowing the X and Z axes of the joint to rotate aswell (`<axis xyz="1 1 1"/>`).
I've found [this forum post](https://pybullet.org/Bullet/phpBB3/viewtopic.php?t=13035) of someone achieving a more rope-like behavior, but they have issues with the stability of it.
What do you think?

jet-sony, 2024-08-05:
I would argue for using the URDF file since it's a better configuration method, the same things can be achieved.

I also wouldn't do axis=(1,1,1). This actually makes it rotate around an offset axis with a vector of x=1 y=1 z=1. Instead, you'll want to make very short links and alternate between (1,0,0), (0,1,0) and (0,0,1). Perhaps it it stiff because of the damping parameter, I will play with it between today and tomorrow and see how I can get the links to move freely.

You'll have to define the many links manually yourself in the URDF tho, I'm too lazy to fopy and paste it and shift coordinates at this time 🫢.

jet-sony, 2024-08-05:
Another alternative to have true ball joints is maybe use an SDF file.

jjshoots, 2024-08-05:
@defrag-bambino After messing around with it for a bit, this seems to work. Essentially, I approximate the string as a 0 mass, 0 collision entity, with a maximum extension length. This allows us to approximate the string as a prismatic joint, but careful attention must be made to ensure that the anchor and the ball both have negligible moments of inertia. Here's what worked for me:

**ball_and_string.urdf**
```xml
<?xml version="1.0"?>
<robot name="ball_and_string">

  <!-- Ball link -->
  <link name="ball">
    <visual>
      <geometry>
        <sphere radius="0.1"/>
      </geometry>
      <material name="black">
        <color rgba="0.0 0.0 0.0 1.0"/>
      </material>
    </visual>
    <collision>
      <geometry>
        <sphere radius="0.1"/>
      </geometry>
    </collision>
    <inertial>
      <mass value="0.1"/>
      <inertia ixx="0.000001" ixy="0.0" ixz="0.0" iyy="0.000001" iyz="0.0" izz="0.000001"/>
    </inertial>
  </link>

  <!-- Anchor link -->
  <link name="anchor">
    <visual>
      <geometry>
        <box size="0.03 0.03 0.03"/>
      </geometry>
      <material name="blue">
        <color rgba="0.0 0.0 1.0 1.0"/>
      </material>
    </visual>
    <inertial>
      <mass value="0.001"/>
      <inertia ixx="0.000001" ixy="0.0" ixz="0.0" iyy="0.000001" iyz="0.0" izz="0.000001"/>
    </inertial>
  </link>

  <!-- Sliding joint -->
  <joint name="ball_to_base" type="prismatic">
    <parent link="ball"/>
    <child link="anchor"/>
    <origin xyz="0 0 0.5" rpy="0 0 0"/>
    <limit lower="0.0" upper="1.0" effort="0.0"/>
    <axis xyz="0 0 1"/>
  </joint>

</robot>
```

**main.py**
```py
import numpy as np

from PyFlyt.core import Aviary

# the starting position and orientations
start_pos = np.array([[0.0, 0.0, 1.5]])
start_orn = np.array([[0.0, 0.0, 0.0]])

# environment setup
env = Aviary(
    start_pos=start_pos,
    start_orn=start_orn,
    render=True,
    drone_type="quadx",
    drone_options=dict(
        drone_model="primitive_drone",
        use_camera=True,
        camera_fps=30,
    ),
)

# spawn in the pendulum
pendulum_id = env.loadURDF(
    "./ball_and_string.urdf",
    basePosition=np.array([0.0, 0.0, 1.0]), # start drone pos - start string length
    useFixedBase=False,
    globalScaling=1.0,
)

# register all new bodies with Aviary
env.register_all_new_bodies()

# create a constraint between the base of the pendulum object and the drone
env.createConstraint(
    env.drones[0].Id,
    -1,
    pendulum_id,
    0,
    env.JOINT_POINT2POINT,
    np.array([0.0, 0.0, 0.0]),
    np.array([0.0, 0.0, 0.0]),
    np.array([0.0, 0.0, 0.0]),
)

# disable motor on the prismatic joint
env.setJointMotorControl2(
    pendulum_id,
    0,
    env.VELOCITY_CONTROL,
    force=0,
)

# simulate
env.set_mode(7)
env.set_setpoint(0, np.array([0.0, 0.0, 0.0, 3.0]))
for i in range(5000):
    env.step()

```

defrag-bambino, 2024-08-14:
Thanks!
I've made this into a gym env. Here's my fork: [https://github.com/defrag-bambino/PyFlyt-BallInCup](https://github.com/defrag-bambino/PyFlyt-BallInCup).
I guess the reward function is subject to change, as it probably is not optimal yet. Until now I've only managed to train an agent that swings the ball around itself to gather reward. I am not sure if the "rope" is even capable of properly reducing its length, i.e. allowing the ball to land exactly on top of the drone.

defrag-bambino, 2024-08-15:
There is this Deepmind Control Suite env 2D "[ball in cup](https://github.com/google-deepmind/dm_control/blob/main/dm_control/suite/ball_in_cup.xml)", which uses tendons as the joint. But it seems like URDF does not support this. However, the MJCF format does and it [seems](https://github.com/search?q=repo%3Abulletphysics%2Fbullet3%20tendon&type=code) PyBullet can load this aswell?

defrag-bambino, 2024-08-15:
Ok, I've updated the reward function. It manages to swing the ball onto the drone's top. However, it seems to collide with something invisible there? Maybe I'll have to add a "catch area" of sorts.

jjshoots, 2024-08-15:
This looks really good, if you could make a PR when it's ready, I could pick it up, clean it up, and make it official.

jjshoots, 2024-08-15:
Could you upload a video here of:
1. Working swing up
2. Collision with the invisible thing

defrag-bambino, 2024-08-16:
Here's one with a smaller ball
[Screencast from 16.08.2024 10:08:32.webm](https://github.com/user-attachments/assets/84bca353-6ec8-4a78-8d49-aafe285cfe3c)

jjshoots, 2024-08-18:
@defrag-bambino could you make a PR of your branch to the `ball_in_cup` branch? I'd like to test run your environment using my learning stack

closed 2024-09-04 by jjshoots