jjshoots/PyFlyt #61: Model output to Crazyflie Control

zcase, 2024-09-20:
Hi great repo! Do you have an example using the stablebaselines3 algorithms such as PPO that controls an actual crazyflie drone? I just trained a model using the (vp, vq, vr, T) flight mode and would like to integrate into a crazyflie. I was just curious if you already had a working example.

jjshoots, 2024-09-21:
Hi! Thanks for dropping by.

I don't have an end-to-end sim-train-to-real example, but I have code that shows how to interact with the crazyflie using setpoints [here](https://github.com/jjshoots/CrazyFlyt/blob/master/examples/sim_n_fly_single.py).

The code is fairly undocumented as of now, so feel free to leave this issue open and leave questions if you want.

jet-sony, 2024-09-24:
Yep, both of those assumptions are correct. Positions are always in global while velocities are local.

zcase, 2024-10-04:
sweet thank you,

When using the default 0 mode with angular velocities, based off pybullet having things in radians, I assume the first 3 are in radians and is the T in percentage?

jjshoots, 2024-10-05:
Yep, that's correct. More concisely, T is in ratio, ie: 1.0 is 100% and 0.0 is 0%.

zcase, 2024-10-07:
@jjshoots so does 0 mean hover thrust or no motor power?

jjshoots, 2024-10-08:
0 means motor stop. You can actually input negative thrust too, in which case, the motor will spin in reverse and force and torques will be reversed. This allows us to simulate 3D drones.

However, for the current quadx API, there is a motor min-throttle clip here: https://github.com/jjshoots/PyFlyt/blob/master/PyFlyt/core/drones/quadx.py#L493

zcase, 2024-10-08:
@jjshoots ok that makes sense. I just wanted to verify.

Now is there a way to vary the mass of the drone during reset? I can't seem to find anything that loads in the urdf mass for something like the hover training.

jjshoots, 2024-10-09:
Masses are defined in the URDF by default in pybullet. But if you want to change it during runtime, you could do something like this: https://github.com/jjshoots/PyFlyt/blob/master/PyFlyt/core/abstractions/boosters.py#L207

zcase, 2024-10-28:
@jjshoots SO I tried your CrazyFlyt and when I connected it to the hardware with the flowdeck it just flipped on the ground and didn't really take off at all. I couldn't even get it to hover with the simple example you have. Did you do anything special to get it to work with the real hardware?

jjshoots, 2024-10-29:
I assume you have tried just hovering normally and verifying that it works that way? There are a number of things that could cause it to just flip, common examples are flipped props, reversed sensors, etc. I would try doing a normal flight using simple bluetooth control via the crazyflie's official implementation first.