jjshoots/PyFlyt #63: Clarification of Coordinate Systems in Documentation

defrag-bambino, 2024-10-01:
Hi,

I am trying to deploy a learned controller from PyFlyt onto a real quadcopter running Ardupilot. As far as I can tell these two use different coordinate system conventions. Namely: PyFlyt uses ENU (which means on the body of the drone its forward is X, to the left is Y and up is Z; Ardupilot uses NED (i.e. forward X, right Y and Z down). Is this correct thus far?
Also, I am unsure about the direction of rotation for yaw pitch roll in PyFlyt. Looking at the illustration provided in the paper, which way is a positive roll for example?
[image]
Ideally, the image containing this information should provided in the documentation.

Thanks!

jet-sony, 2024-10-01:
Hi, admitttedly, even I had an issue with different axis convention when trying to deploy agents onto real quadcopters (PX4, Crazyflie, and Ardupilot all use different conventions).

PyFlyt used ENU / FLU:
- Front X
- Left Y
- Up Z

Ardupilot uses NED / FRD:
- Front X
- Right Y
- Down Z

Rotation conventions always follow right hand thumb rule, this is the standard regardless of NED/ENU:
[image]

Therefore, for PyFlyt (Front X) and Ardupilot (Front X), roll positive would be facing-forwards-clockwise.
However, in PyFlyt (Left Y), pitching positive would be pitching forwards; in Ardupilot (Right Y), pitching positive would be pitching backwards.

Hope this clarifies things!

defrag-bambino, 2024-10-04:
Thanks!
What is the range for the rotations? Are they in [0, pi] or [-pi/2, pi/2] or [0,1] ...?

jjshoots, 2024-10-04:
[-pi, pi].

defrag-bambino, 2024-10-07:
And for angular velocities? Rad/s?

jet-sony, 2024-10-07:
Yep!

defrag-bambino, 2024-10-10:
I think I've found my issue. It is related to the network outputs, rather than inputs.
The flight mode documentation says "0: vp, vq, vr, T" (angular velocities). Therefore I expected the four outputs of the network to be in the order "pitch, yaw, roll, throttle". However, it seems that it is actually "yaw, pitch, roll, throttle"?

jet-sony, 2024-10-11:
Are you intending to output to a crazyflie IRL? If the policy was trained in one of the gym envs within PyFlyt, the outputs should be "roll, pitch, yaw, theust" (v-pqr, T). The rotational outputs always follow the axes of the linear frames. If your intended goal is to output to a crazyflie, then it is likely in the form of "pitch, roll, yaw, thrust". I would try isolating outputs (by setting everything else to 0), and then activating only one output.

zcase, 2024-10-28:
@jet-sony just to clarify what you said above, are you saying that if we trained a model within the environment the model should output int the order of "roll, pitch, yaw, thrust" and so if i take this model and try to control an actual crazyflie drone i should change the order from "roll, pitch, yaw, thrust" to "pitch, roll, yaw, thrust"?

jjshoots, 2024-10-29:
@zcase it's been awhile since I've messed with crazyflies. What I will say is that the output space in PyFlyt is permuted in Crazyflies. I would refer to [the official documentation](https://www.bitcraze.io/documentation/repository/crazyflie-lib-python/master/user-guides/python_api/) for the most up to date material. CrazyFlyt is simply a simple wrapper around both these libraries.