jjshoots/PyFlyt #64: Rocket Landing Task Fail

Pytha11, 2024-11-16:
I have tried some RL algorithms, but all of them do not success to land.
Do you have a successful case? Like, use PPO with some specific hyper-parameters.

jjshoots, 2024-11-16:
I never actually expected anyone to try the rocket landing environment, and I have only gotten it to work in some very rare edge cases.

So TLDR; it's actually not solved. I would be very happy if you want to take over designing a reward structure/sensor scheme for the environment. I can put you on the webpage when it is solved and also acknowledge your contributions in the paper when it is published.

Pytha11, 2024-11-16:
I would like to contribute to the code, but I need more time to understand the physical dynamics. I think it is still a challenge.
For example, I only allow the rocket to use booster ignition and throttle, like action = [0, 0, 0, 1, 1, 0, 0], but I got speed on x and y, like linear velocity = [0.3, 0.4, 0.5] instead of [0, 0, 0.5].
I think I cannot find the detail in your published paper.

jjshoots, 2024-11-17:
By default, you will not be able to land the rocket with only vertical actuators (only throttle nd ignition) since the rocket starts in an arbitrary rotated orientation and there is also output noise. You can disable those here:
- [Set `randomize_drop=False` here](https://github.com/jjshoots/PyFlyt/blob/master/PyFlyt/gym_envs/rocket_envs/rocket_landing_env.py#L100)
- [Set `noise_ratio=0` here](https://github.com/jjshoots/PyFlyt/blob/master/PyFlyt/models/vehicles/rocket/rocket.yaml#L19)
- [Pin the landing pad position here](https://github.com/jjshoots/PyFlyt/blob/master/PyFlyt/gym_envs/rocket_envs/rocket_landing_env.py#L116-L120)

There may still be some level of noise in the system, but disabling those settings should make things much easier.

If you'd like to start tweaking the reward structure, it is defined [here](https://github.com/jjshoots/PyFlyt/blob/master/PyFlyt/gym_envs/rocket_envs/rocket_landing_env.py#L192).

Pytha11, 2024-11-19:
I split the landing into two simple tasks to help me find the potential reward design.
vertical landing                      and        "horizontal" landing 
        o (start position)                                   o 
        *                                                                  *
        *                                                                        *
       O (target position)                                                    O
Could you help me to give some demonstrations of action for agent with expert knowledge on how to successfully land in the "horizontal" landing?

jjshoots, 2024-11-26:
Hi, sorry for the super late reply. I don't actually have demonstrations of a successful horizontal landing, but I would imagine that some level of gliding is required. The rocket body is represented as a `boring_body`, and hence can actually produce lift/drag depending on orientation.

jjshoots, 2024-11-28:
Do you happen to have a video of the landing?

Pytha11, 2024-11-30:
Uploading Bullet Physics ExampleBrowser using OpenGL3+ [btgl] Release build 2024-11-27 11-53-03.mp4…

[image]
[image]

jet-sony, 2024-11-30:
First off, that video and the graph looks amazing. For the issue of the algorithm exploiting the velocity reward when away from the pad, one solution I can imagine is slightly increasing the horizontal offset penalty, and increase the final landing reward slightly. We essentially need to encourage the agent that high offset is a bad thing. I would actually propose using a binary constant penalty for this, induced when the rocket is x distance away from the pad (probably x = pad radius).

Pytha11, 2024-12-02:
I only disable your rewards for each step and add a negative abs(speed at z axis) as a reward for each terminal of an episode (like sparse reward). Moreover, I only use booster ignition and throttle for the vertical landing task. It makes rocket landing as a simple task so that PPO can find out the appropriate solution. For landing with other conditions, I still fail to make it.

jjshoots, 2024-12-15:
I have fixed the environment in #68, it should be learnable now. I have also made a new release with this fix.

closed 2024-12-15 by jjshoots