jjshoots/PyFlyt #65: Fixed-wing waypoints

tlaurie99, 2024-11-25:
Hello and thank you for your repo -- I have been using the quadcopter waypoint environment and the fixed-wing dogfighting environment (an older version I have modified). I am now interested in the fixed-wing waypoints environment and I am finding that I cannot get the aircraft to go toward any waypoint. Even while changing the `goal_reach_distance ` to really large, the aircraft will still just fly around randomly. I've tried various PPO implementations (distributional value functions, exploration-based critics, etc.), but cannot seem to get the agent to find a waypoint like the quadcopter environment. Typically, I can get the quad to reach all 4 waypoints within ~1M timesteps, but the fixed-wing has been well over 100M with no success. Do you have any recommendation what this might be from or what I might be doing incorrectly? I will have the config below for reference with hypers. 

```
'''Set the environment config -- we can pass the waypoints to train on here'''
env_config = {
    'sparse_reward': False,
    'num_targets':  4,
    'goal_reach_distance': 5.0,
    'flight_mode': 0,
    'flight_dome_size': 100.0,
    'max_duration_seconds': 120.0,
    'angle_representation': "quaternion",
    'agent_hz': 30,
    'render_mode': None,
    'render_resolution': (480, 480),
    # 'landing_waypoints': [[1000, 52.4078, 0], [400, 20.9631, 0], [100, 5.2408, 0], [0, 0, 0]],
}

config = PPOConfig().training(
    gamma = 0.99,
    lambda_ = 0.95,
    # kl_coeff = 0.5,
    num_sgd_iter = 15,
    lr_schedule = [[0, 0.0003], [15_000_000, 0.00025], [30_000_000, 0.0002], [50_000_000, 0.0001]],
    vf_loss_coeff = 0.5,
    # vf_clip_param = 15.0,
    clip_param = 0.2,
    grad_clip_by ='norm', 
    train_batch_size = 100_000, 
    #new api removes "sgd" and is just minibatch_size
    minibatch_size = 20_000,
    grad_clip = 0.5,
    model = {'custom_model': 'ClampedCritic', 
           'vf_share_layers': False,
           'fcnet_hiddens': [256,256],
           'fcnet_activation': 'LeakyReLU',
            }
).environment(env = 'WaypointsCustomEnv-v0', env_config = env_config
).rollouts(
num_rollout_workers = 28
).resources(num_gpus = 1
)

algo = config.build()

num_iterations = 500
results = []

for i in range(num_iterations):
    result = algo.train()
    print(f"Iteration: {i}, Mean Reward: {result['env_runners']['episode_reward_mean']}")
    results.append([result['env_runners']['episode_reward_mean'], result['env_runners']['episode_len_mean']])

results_df = pd.DataFrame(results)
    
ray.shutdown()
```

jjshoots, 2024-11-26:
Thanks for bringing this up! I think something is definitely wrong, 100M steps is a lot! I'll look at this later but pinging here just to keep you updated.

tlaurie99, 2024-11-26:
So for whatever reason, it looks like the self.sparse_reward flag was being set to a dictionary, so the agent was only ever seeing -0.1 for each timestep. I changed [this](https://github.com/jjshoots/PyFlyt/blob/e1b47e51dffecb7788ee510207e98439a263e246/PyFlyt/gym_envs/fixedwing_envs/fixedwing_waypoints_env.py#L102) to `False` as a means of overriding and now it is accessing the the non-sparse rewards. Not sure if I had inherently done this on my end, but this might be a source for you to glance at to see if it works as intended for you.
```
sparse_reward inside compute_term_trunc_reward: {'sparse_reward': False, 'num_targets': 1, 'goal_reach_distance': 5.0, 'flight_mode': 0, 'flight_dome_size': 20.0, 'max_duration_seconds': 120.0, 'angle_representation': 'quaternion', 'agent_hz': 30, 'render_mode': None, 'render_resolution': (480, 480), worker=1/5, vector_idx=0, remote=False}
```
This definitely helped as now the agent learns within a few million timesteps. Thanks!

tlaurie99, 2024-11-26:
update: this does indeed seem to be my fault! I forgot to unpack the config args `**config` while registering the env. with RLLIB. My apologies!

(If future folks have an issue I had):
```
def env_creator(config):
    env = FixedwingWaypointsEnv(config)
    env = FlattenWaypointEnv(env)
    return env
register_env('FixedwingWaypointsEnv-v0', env_creator)
```
vs.
```
def env_creator(config):
    env = FixedwingWaypointsEnv(**config)
    env = FlattenWaypointEnv(env)
    return env
register_env('FixedwingWaypointsEnv-v0', env_creator)
```

closed 2024-11-26 by tlaurie99