jjshoots/PyFlyt #66: Velocity vectors not pointing along heading

felix-basiliskroko, 2024-12-10:
For my Bachelorthesis I am doing a review of RL-Algorithms for UAV Waypoint Navigation and so I was looking into your [QuadXWaypointsEnv](https://github.com/jjshoots/PyFlyt/blob/master/PyFlyt/gym_envs/quadx_envs/quadx_waypoints_env.py) environment. My reward function computes the azimuth- and pitch-angle offset between the UAVs heading (i.e its velocity) and the Line-of-Sight (LOS) between its position and the targets position and after some transforming it would gain maximal reward if its velocity vector is perfectly aligned with the LOS indicating direct heading towards the target. 
To debug I plotted the linear velocity vectors at each timestep and noticed this:

[image]

The velocities were always pointing away from the actual direction in which the Quad was moving to and I am failing to make logical sense out of this. I hope you can help me with this one, even if it might show some lacking of fundamental physical concepts.

PS: Is there a way of directly accessing the UAVs acceleration as well as some acceleration constraints? Since I use angular positions as control inputs I would want to have an action-validation in the background to ensure pitch-, yaw-, roll-adjustments won't cause instabilities in the UAVs movement. 
Thanks a lot for your help

jjshoots, 2024-12-11:
Hi! Thanks for raising the issue, could you deacribe a bit more what I am looking at in the photo, and how you are accessing the linear velocity component?

For acceleration, there is no direct way, but you could possibly try taking the discrete difference of velocity between timesteps to obtain acceleration.

felix-basiliskroko, 2024-12-11:
Hey, what you see in the picture is the linear velocity of the UAV at each timestep. I access the linear velocity using the `compute_attitude` function of the [QuadXBaseEnv]()-class. And thanks for the info about the acceleration!

felix-basiliskroko, 2024-12-11:
Perfect. That helps me alot, will do that. One more thing was if there is some acceleration constraints implemented by PyBullet for the Quadcopter? Thanks alot for your help!

jjshoots, 2024-12-11:
I don't think there are any acceleration constraints. At the very least, this was not implemented in PyFlyt, but PyBullet (the underlying physics engine) may have it implemented. I highly doubt so, though.

felix-basiliskroko, 2024-12-12:
Good to know, thanks a lot for your help!

closed 2024-12-12 by felix-basiliskroko