jjshoots/PyFlyt #67: Modeling New Quads

zcase, 2024-12-13:
If I wanted to model another quadcopter that is not a crazyflie. What are the places in the base Quadx code that would need to be updated or a potential gotcha when modeling a new quad? OR can I just do a new urdf and config and we should be good?

jjshoots, 2024-12-15:
Hi, thanks for the issue and apologies for the late reply.

You should be able to just use a new urdf and config.

Here's an example:

Your expected file structure:
```
/your_drone_folder
  /your_drone_name
    your_drone_name.urdf
    your_drone_name.yaml
```

The code to spawn the drone into the Aviary:
```py
env = Aviary(
    start_pos=start_pos,
    start_orn=start_orn,
    drone_type="quadx",
    drone_options={"model_dir": "/path/to/your_drone_folder", "drone_model": "your_drone_name"},
)
```

That said, the urdf and yaml files are expected to be in the same form as that of the crazyflie. You can see what they look like [here](https://github.com/jjshoots/PyFlyt/tree/master/PyFlyt/models/vehicles/cf2x).

zcase, 2024-12-16:
@jjshoots Thank you for your response. Does anything have to happen with the motor mapping? I know the crazyfile have an unconventional yaw pitch and roll (i think the its the roll that has the left hand rule instead of the right hand rule but everything else is the same) Does that have to be taken account of or by just doing the folder structure and urdf files map that into account?

jjshoots, 2024-12-17:
As long as the motor ids in the urdf are similar to the pre-implemented one, it should be fine. We actually use the same convention as the PX4 one.

zcase, 2024-12-17:
@jjshoots just to clarify your saying that as long as they are in the the same order like the comment says here:

```
        DRONE CONTROL
            motor ids correspond to quadrotor X in PX4,
            -------------------------
            |  (cw)2  0(ccw)        |
            |      \\//           x |
            |      //\\         y_| |
            | (ccw)1  3(cw)         |
            -------------------------
            using the ENU convention
            control commands are in the form of roll-pitch-yaw-thrust
        """
```

jjshoots, 2024-12-17:
That is correct. :)