jjshoots/PyFlyt #71: The camera in the example is not updated

AmiyaShirou, 2024-12-19:
When I checked the parameters of aviary, I found that the parameters of camera were not needed, but 04_camera.py in the example still existed
Is there any simple and clear example of the new version?
ty

jjshoots, 2024-12-19:
Thanks for the issue! The parameters are being passed into `drone_options` of the Aviary, which is passed into the drones themselves. Does this answer your query?

AmiyaShirou, 2024-12-19:
Thanks for your reply, now I know the parameters are in drone_options, but when I run
_**env = Aviary(
    start_pos=start_pos,
    start_orn=start_orn,
    render=True,
    drone_type="quadx",
    drone_options=dict(
        use_camera=True,
        camera_angle=30,
        camera_FOV_degrees=110,
        camera_fps=30,
    ),**_           
i cant send the parameters into Aviary
TypeError: QuadX.__init__() got an unexpected keyword argument 'camera_angle'
Maybe I'm too dumb to figure it out on my own.If possible,could u teach me how can i solve the problem?

jjshoots, 2024-12-19:
The parameter is `camera_angle_degrees` according to [here](https://github.com/jjshoots/PyFlyt/blob/master/PyFlyt/core/drones/quadx.py#L34). :)

anastasi0s, 2025-03-22:
> The parameter is `camera_angle_degrees` according to [here](https://github.com/jjshoots/PyFlyt/blob/master/PyFlyt/core/drones/quadx.py#L34). :)

in the parameter in the 04_camera.py  example is still `camera_angle` instead of `camera_angle_degrees`