jjshoots/PyFlyt #76: Fixe-wing Model

ZF113120, 2025-02-25:
Hello! Thank you very much to you and your team for developing PyFlyt. I have a small request, could you please tell us the performance parameters of the fixed-wing aircraft in PyFlyt? For example, the stall speed, mass, etc.? In addition, does PyFlyt accept aircraft models from other high-fidelity aircraft simulators?

jjshoots, 2025-02-27:
Hi! Thanks for the question. All parameters of the fixedwing model are defined here.

https://github.com/jjshoots/PyFlyt/blob/master/PyFlyt/models/vehicles/fixedwing/fixedwing.yaml

Yes, you can edit these parameters and even edit the geometry of the aircraft, but I'm not sure how feasible it is to import models from other simulators.

ZF113120, 2025-03-04:
Thank you for your assistance. I have a small suggestion: it would be beneficial to include some examples of RL controlling UAV in the project. This would help everyone learn how to use PyFlyt more quickly.